rusefi/alphax-8chan #3: analog mux

rusefillc, 2022-06-28:
https://github.com/rusefi/stim is finally a success shall we use same analog mux chip on 8chan for VBatt/CLT/IAT/something else?

mck1117, 2022-06-28:
Use the same 2-input muxes as inside the VR-discrete module.

Usable on any channel other than MAP or knock.

rusefillc, 2022-06-28:
[image]

C87403
NC7SB3157P6X $5 no way

@andreika-git did we substitute on alphax?

mck1117, 2022-06-29:
there are plenty of cheap options like  C91307

https://jlcpcb.com/parts/componentSearch?isSecondLevel=true&name=Analog%20Switches%20%2F%20Multiplexers%20&firstName=Interface%20Ics

rusefillc, 2022-10-01:
#17

ElDominio, 2022-10-01:
I'd say choose some of the last pins in the pinout to be low priority Analog inputs, maybe have the first 10/12 high priority then the rest low priority?

rusefillc, 2022-10-01:
@andreika-git let's mux one of the aux analog inputs with a 3rd single channel mux controller by same AMUX line

rusefillc, 2022-10-03:
@andreika-git let's mux one of aux from #16

andreika-git, 2022-10-05:
Done!
[image]
[image]

rusefillc, 2022-10-05:
@mck1117 :)

mck1117, 2022-10-05:
@andreika-git wait, is that mux BEFORE the analog input filtering stage? That WILL NOT work. It must be muxed as close to the MCU as possible. The analog input filters only have a few hundred hz of bandwidth, so switching between inputs at hundreds of hz will not work.

mck1117, 2022-10-05:
> @mck1117 what's your level of confidence with single control line, multiple 2-input muxes implementation?

100% confident, but the mux does need to be in the correct spot for it to work.

andreika-git, 2022-10-06:
Now mux is between filtered signals closer to mcu:
[image]

closed 2022-10-07 by andreika-git