rusefi/alphax-8chan #5: configurable high OR low side aux output, 5A

rusefillc, 2022-07-24:


rusefillc, 2022-07-30:
would TLE9201 be the best or at least one of the options for high-current configrable push-pull output?

rusefillc, 2022-08-05:
@dron0gus says MC33879 but 0.6A

Qwerty-OFF, 2022-08-05:
MC33887APVW

rusefillc, 2022-08-06:
MC33887APVW looks great!

@dron0gus can you please double-check just to confirm that we can use MC33887APVW as two independent pull/push outputs?

dron0gus, 2022-08-08:
@rusefillc 
It is possible to use MC33887APVW as two independent push/pull,
BUT it is not possible to independently switch one output to Hi-Z. Only both at the same time.
[image]

dron0gus, 2022-08-08:
TLE94112EL is cool. But almost out of stock.

rusefillc, 2022-09-25:
@mck1117 what are your thoughts on MC33887APVW as most flexible push/pull chip?

closed 2022-10-01 by rusefillc