rusefi/alphax-8chan #7: On-board MAP yes or no

rusefillc, 2022-07-30:
@eldominio if we fancy a fancy CNC metal case do you see 8chan having on-board MAP sensor or not really?

ElDominio, 2022-07-30:
Yes please, most people need it since they don't know how to connect cables.

rusefillc, 2022-07-30:
"Hex Bulkhead Straight Fitting 2PCS Hose Barb Bulkhead Hex Union Brass Barbed Fitting Connector with 1/4inch Flat Washers"

[image]

rusefillc, 2022-07-30:
@ElDominio where do you want to place the nipple? would you like the nipple between 60 and 60 or else?

rusefillc, 2022-07-30:
@ElDominio what sensor do you prefer?

there is MPX5700AP which seems to be more suited for brass fitting and hose without turns inside the ECU

rusefillc, 2022-08-03:
Status update:
@ElDominio uses something barbed on 4chan cases already! for now 8chan should use same exact map as 4chan and @ElDominio plans to use same exact barbed fitting and internal hose solution

ElDominio, 2022-08-03:
https://www.ebay.com/itm/252780287663 This is the item i use, and a hose inside the case

rusefillc, 2022-08-04:
let's have a triple-footprint:

to accomodate both sensors soldered directly to PCB (horizontal 2.5 bar and vertical 4bar)

and also a custom 0.1" strip for angled mount. I guess 0.1" strip could actually match 2.5bar pinout?

andreika-git, 2022-10-01:
[image]

closed 2022-12-11 by andreika-git