rusefi/alphax-8chan #12: MAP/MAF & PPS1/PPS2 clarification

rusefillc, 2022-09-25:
We have

[image]

Where are MAP/MAF & PPS1/PPS2?

ElDominio, 2022-09-25:
None of them are dedicated? All can be used tofr those functions

rusefillc, 2022-09-25:
@ElDominio question about your exact ford motor

rusefillc, 2022-09-25:
At the moment we can only do 80% of thunder. We need to make sure our 80% fits your harness

andreika-git, 2022-10-03:
Currently the following pinout is used:
- PPS1 = pin A33
- PPS2 = pin A14
- MAP1 = pin A22
- MAP2 is used for internal MAP sensor

[image]

rusefillc, 2022-10-20:
@ElDominio confirm that you plan to repin current harness since we have no data on existing harness

ElDominio, 2022-10-20:
[image]
 APS = PPS
 
 
[image]

DI just in case

andreika-git, 2022-10-23:
MAP1 is now on A22
PPS1 = A17
PPS2 = C31

closed 2022-10-23 by andreika-git