rusefi/alphax-8chan #13: DC motor/pull/push TLE9201 proposal

rusefillc, 2022-09-27:
TLE9201 has benefit of availability, known for us part and much cheaper than MC33887APVW as described at https://github.com/rusefi/alphax-8chan/issues/5#issuecomment-1257261740

What can we accomplish if we use four of TLE9201?
I am assuming that each TLE9201 could be one of three usages for us
* legic DC-motor
* two low side channels
* one high side channel, second channel would need to be ununsed if one channel used for high side

If my abovestated assumption is correct it looks like four of TLE9201 is very promising:
"average" user can do dual ETB and two high side channels (one per TLE9201)
most demanding users can do four ETBs
also possible one ETB and three high sides

overall looks like a pretty good set of functionality as long as i understand TLE9201 possibilities correctly

andreika-git, 2022-10-03:
Now we are using 4 H-bridges:
[image]

1) C18+C26 = ETB
2) D20+D21 = DC motor
3) A26+A28 = low/high side "hybrid" outputs (dependent)
4) A27+A29 = low/high side "hybrid" outputs (dependent)
  * A26 and A27 correspond to different H-bridge chips because they are connected for IMRC1 and IMRC2.
 
Currently TLE9201 are used for all 4 chips.

rusefillc, 2022-10-03:
what is IMRC1 ?

also need to move the knowledge to wiki or ideally .pdf

ElDominio, 2022-10-04:
IMRC = Intake Manifold Runner Control, Coyotes have two solenoids (not unlike boost solenoids)  that control some flappers in the intake manifold

rusefillc, 2022-10-05:
@ElDominio please clarify if low side or high side

ElDominio, 2022-10-05:
low side

rusefillc, 2022-10-05:
@andreika-git since low side let's make A26+A27 and A28+A29 pairs to make it a little less confusing

andreika-git, 2022-10-08:
Done!
[image]

closed 2022-10-08 by andreika-git

rusefillc, 2023-01-22:
[image]