rusefi/alphax-8chan #15: VND7 not VND14NV04-E

rusefillc, 2022-09-30:
@ElDominio please approve VND7 for aux low side outputs to stay real and to get JLC assembly

ElDominio, 2022-09-30:
Only for low side correct? Injectors still VND14? if so that's fine

rusefillc, 2022-10-01:
See also #18

rusefillc, 2022-10-02:
@ElDominio I need help understanding why do we need VND14 for injectors?

Actually the plan was to use SOT223 injector drivers like ZXMS6005SGTA for injectors same as hellen154hyundai or nissan

we would need to revisit if we have desire to support v10 i.e. if we need up to 10 injectors with same MOSFET

but first thing first: why VND14 for injectors?

rusefillc, 2022-10-03:
@ElDominio was confused

got confirmation over phone no need for VND14 anywhere

rusefillc, 2022-10-03:
For 8 injectors we will use the output module as usual

@andreika-git please fix schematics to say VND7 not VND14

andreika-git, 2022-10-05:
[image]

closed 2022-10-05 by andreika-git