rusefi/alphax-8chan #16: How many analog inputs do we have?

rusefillc, 2022-09-30:
Looks like we are low on analog inputs?

[image]

I assume a lot of A/D decisions are consequences of 5 trigger input requirment?

Shall we some of not-input-module pins for trigger input in order to get more analog inputs?

See also https://github.com/andreika-git/hellen-one/issues/155

ElDominio, 2022-09-30:
"shall we some of" what?

I was going to propose, I have heard of the "rolling anlog inputs" that have been proposed (I think one processor pin reads multiple analog inputs?)

Can we then have high/low priority input pins? For example, CLT/IAT is know to not read TOO MANY times per second, so assign that to low priority analog inputs, then critical stuff like wideband/MAP/TPS/fuel pressure/oil pressure have a dedicated analog input pin?

andreika-git, 2022-10-03:
Added 3 more analog inputs:
[image]
They were placed on C31..33 connector pins:
[image]

Now we have 14 analog inputs and 14 digital inputs total.

The crank input was moved to the PA10 pin USBID (5V-tolerant):
[image]

Wakeup input was moved to the PE0 pin (UART8_RX) and changed to 3.3v level:
[image]
[image]

Also mcu144-0.7 revision is now used.

rusefillc, 2022-10-04:
@andreika-git please clarify what MCU has PE0 as WKUP or how would wake up work overall? I see only PA0 for 407 and 427?

andreika-git, 2022-10-05:
The WKUP pin is on PA0:
[image]

Currently we still have +4 free analog inputs (2 muxed), because 2STEP is now on PE0:
[image]