rusefi/alphax-8chan #19: sleep mode current consumption

rusefillc, 2022-10-04:
what can we do for proper low current sleep mode?

rusefillc, 2022-10-04:
See also https://github.com/rusefi/alphax-4chan/issues/65

andreika-git, 2022-10-08:
1. Added a switchable 5V digital power supply:

[image]

2. Added a switchable 5V analog power supply (based on TLS115):

[image]

3. Changed some consumers to use switchable power:
- sd-card (3.3v)
- internal map sensor (5va)
- 2 knock modules (5va)
- 3 muxes (5va)
- 2 discrete-vr modules (5v)
- some leds (5v)

rusefillc, 2022-10-20:
@andreika-git are you ready to apply label:"schematics: DONE"?

rusefillc, 2022-11-30:
@Qwerty-OFF please review

andreika-git, 2022-12-11:
Still need more tests... + Waiting for megamodule

closed 2022-12-11 by andreika-git