rusefi/alphax-8chan #22: What is IN_FLEX

rusefillc, 2022-10-05:
@ElDominio flex sensor input is it always digital or is it sometimes analog?

rusefillc, 2022-10-05:
https://github.com/andreika-git/hellen-one/issues/155

ElDominio, 2022-10-05:
Even though it is sometimes analog, if I read "IN_FLEX" I automatically assume it is a digital input. Anyone that wants to use an analog (puke) flex sensor should use a low speed Analog In

rusefillc, 2022-10-05:
For 8chan Flex is digital only

andreika-git, 2022-10-08:
This issue is dependent on https://github.com/andreika-git/hellen-one/issues/171
Currently we have no more free pins to move flex input anywhere.

rusefillc, 2022-10-20:
Progress on https://github.com/andreika-git/hellen-one/issues/171 we have free pins now!

andreika-git, 2022-10-23:
Moved IN_FLEX and CAM_HALL to separate digital input stages:

[image]

Moved FLEX to another digital pin:

[image]

Which allowed to add two more analog inputs (muxed)!

[image]
[image]

closed 2022-10-23 by andreika-git