rusefi/alphax-8chan #24: Schematics is DONE need review

rusefillc, 2022-10-23:
This https://github.com/rusefi/alphax-8chan is now SCHEMATICS DONE

rusefillc, 2022-10-23:
@Qwerty-OFF can you please review?

@mck1117 can you please review?

mck1117, 2022-10-23:
Add a 1n cap from the input and after the resistor to ground for the digital inputs. Better noise suppression.

Add here (and to others):

Add from `IN_CAM_HALL` and pin 3 to ground (same to all channels).

[image]

Proteus for reference:

[image]

mck1117, 2022-10-23:
Connect LED to processed input, not raw input for `IN_CAM_HALL` (others seem to already do this, but not cam??)

[image]

mck1117, 2022-10-23:
Inconsistent muxed analog naming:

It's confusing that some mux [AT1, AT3] -> `IAT`, but others mux to a net called `AN3_4`. The net should not be called `IAT`, because it isn't IAT.

[image]

rusefillc, 2022-10-23:
@mck1117 let's create separate tickets like https://github.com/rusefi/alphax-8chan/issues/25?

andreika-git, 2022-10-25:
#25, #28 and #29 done.

rusefillc, 2022-11-30:
@mck1117 we now have PCB, could be 48 to placing the order

rusefillc, 2022-12-11:
@mck1117 
@dron0gus 
@Qwerty-OFF 

Let's have the last round of review?

closed 2022-12-17 by rusefillc