rusefi/alphax-8chan #30: 0603-4 or x4 0201?

rusefillc, 2022-10-26:
https://github.com/rusefi/alphax-4chan/issues/71 freaks me out regarding R0603-4 availability

also I hear those are not easy to hand-solder if repairs are needed

@andreika-git @mck1117 @Qwerty-OFF @dron0gus what are the pros and cons between 0603-4 array versus four single 0201 components?

Qwerty-OFF, 2022-10-26:
0201 almost impossible to hand-solder. Four 0402 is the preferred option. Also, JLC seems to have a lot of C425239
[image]

mck1117, 2022-10-26:
yeah, please do not use 0201, I'm not even sure JLC will reliably mount those.

0402 is plenty reliable, proteus uses a million of em

rusefillc, 2022-10-26:
It was a typo, I was thinking 4x0402

rusefillc, 2022-10-26:
@Qwerty-OFF @mck1117 @andreika-git what about 0x4 0402?

closed 2022-11-30 by rusefillc