rusefi/alphax-8chan #34: Let's try double-sided WBO and double-sided VR

rusefillc, 2022-11-30:
Since we have two WBO and two VR modules can we use latest double-sided version for one WBO and one VR?

rusefillc, 2022-11-30:
@andreika-git I am sorry not great timing I should have suggested this some time ago :(

mck1117, 2022-12-01:
do we really want to do this right before fab on rev A, or should it wait for rev B? There will surely be a rev B.

rusefillc, 2022-12-01:
@mck1117 there are six changes currently in the works
8chan "ready"
new WBO ready
new VR ready
megamodule _soon_
new power supply module _soon_
new knock _soon_

rev A is a prototype anyway. Since rev A is already a prototype I think it's best to add two more small risks into it.

ElDominio, 2022-12-01:
I said OK to WBO and VR since they are not "mission critical" and i'm getting maybe 1 since 4 will probably be spared around for testing

rusefillc, 2022-12-01:
actually https://github.com/andreika-git/hellen-one/tree/master/modules/knock/0.2 we also have already

mck1117, 2022-12-01:
is there a changelog for modules? what's new in knock 2?

rusefillc, 2022-12-01:
@mck1117 https://github.com/andreika-git/hellen-one/blob/master/CHANGELOG.md