rusefi/alphax-8chan #35: tle9201 not rendered on ibom

rusefillc, 2022-12-01:
[image]

rusefillc, 2022-12-09:
Seems to still be an issue, is this easy or not?

battery seems to be another broken one, what is the level of effort?

rusefillc, 2023-02-01:
@Qwerty-OFF says in relation to ANOTHER board, not sure if relevant or not

"названия у футпринта в хелленне и кикаде разные. Название футпринта в кикаде поменялось когда я в него железно прикрутил большие дырки для пайки"

andreika-git, 2023-02-13:
The problem is in the definition of Infineon_PG-DSO-12-11.kicad_mod. It used a newer Kicad multiline format instead of older one-line format. I've updated the Hellen-One scripts to understand the new format.

closed 2023-02-13 by andreika-git