rusefi/alphax-8chan #36: BOM rev A

rusefillc, 2022-12-01:
Q600,Q600_...Q600,Q600_2 | BTS3028 | DPAK | BTS3028SDRC112639

U103 | AP2331W-7 | SOT-23

R16 | 1k | R0603-4 | CN1J4TTD102JC240722 C240722 hello again

Q6,Q7,Q8,Q...Q6,Q7,Q8,Q9,Q22,Q23,Q26,Q28,Q29,Q30 | VND14NV04TR-E | DPAK | VND14NV04TR-E

rusefillc, 2022-12-01:
@andreika-git need help

I've made https://github.com/rusefi/alphax-8chan/commit/12f094592ebe8c62e7420f8a916011bc03c3b6e9 change which has produced expected output for Q600

but U103 was not removed and Q6 was not replaced :( what do I do wrong?

rusefillc, 2022-12-09:
@andreika-git fresh BOM issues

R16 | 1k | R0603-4 | CN1J4TTD102J

Q6,Q7,Q8,Q... | VND14NV04TR-E

andreika-git, 2022-12-11:
I think I added all components I knew and we're ready to test the BOM on JLC again.

closed 2022-12-17 by Qwerty-OFF