rusefi/alphax-8chan #49: VND10 with 3.3 drive

rusefillc, 2022-12-13:
See https://github.com/andreika-git/hellen-one/issues/236

How is VND10 in this regard?

rusefillc, 2022-12-13:
@Qwerty-OFF says VND10 has the similar curve

mck1117, 2022-12-13:
exact PN?

rusefillc, 2022-12-13:
@Qwerty-OFF stm32 is allowed to output 2.9v?

[image]

rusefillc, 2022-12-13:
> exact PN?

VND10N06TR-E

mck1117, 2022-12-13:
I think VND10xxxx will require 5v drive.

> stm32 is allowed to output 2.9v?

You've read that slightly wrong, `vdd - 0.4` is for supply voltage to the stm32 between 2-2.7 volts. If you're sourcing 20mA with a supply voltage of 2.7-3.6v, it could be as low as `vdd - 1.3`, or in our case, 2.0 volts. BUT - that's when sourcing 20mA, which we aren't trying to do. With the current 4.7k pulldown, it's only 0.7mA, so you'll see very near the full 3.3v.

Qwerty-OFF, 2022-12-17:
Fixed
https://github.com/rusefi/alphax-8chan/issues/52#issuecomment-1356315494

closed 2022-12-17 by Qwerty-OFF