rusefi/alphax-8chan #59: plugs C and D pinout is mirrored

rusefillc, 2023-01-06:
A-B silk and PCB matches pin numbers on the plugs

[image]

C and D silk and PCB does not match numbers on the plugs

[image]

rusefillc, 2023-01-06:
issue is somewhere around https://github.com/rusefi/kicad6-libraries/issues/1 https://github.com/rusefi/alphax-8chan/issues/26 https://github.com/rusefi/alphax-8chan/issues/27

andreika-git, 2023-01-09:
С & D:
kicad6-libraries/Connectors.pretty/6437288-2.kicad_mod 
kicad6-libraries/Connectors.pretty/6473423-1.kicad_mod

rusefillc, 2023-01-17:
So A&B everything as rendered on https://rusefi.com/docs/pinouts/hellen/alphax-8chan/

AT THE MOMENT C&D is mirrored over center line in relation to https://rusefi.com/docs/pinouts/hellen/alphax-8chan/

pin 10 is pin 18
pin 18 is pin 10
pin 8 is pin 26
pin 26 is pin 8

ElDominio, 2023-01-23:
I vote keep mirrored. Gives me peace of mind, and no need for redesign

Qwerty-OFF, 2023-07-20:
@rusefillc что мне сейчас c этим делать?

closed 2023-07-20 by rusefillc