rusefi/alphax-8chan #63: Frequency inputs in plug B all " to be done later"

ElDominio, 2023-01-17:
Not all have to be frequency, some can be simple on/off with a say 10hz refresh rate.
I'd say B14-b19 frequency, B20-b25 10hz on/off digital input

rusefillc, 2023-01-22:
@ElDominio please share more details on what are the actual sensors you are looking to feed into rusEFI

worst case scenario some or _all_ of these would require an additional microcontroller with additional firmware

@mck1117 FYI

ElDominio, 2023-01-23:
assume at least 10x high frequency inputs
-4 vvt
-1 crank
-flex fuel
-4 wheel speeds

then very slow speed switched inputs (5hz is plenty) (can't this be muxed?)
-clutch up
-clutch down
-starter signal
-brake pedal down
-oil pressure switch
-power steering switch
-2x imrc switch
-reverse light switch
about 10x too

what are the DI negatives on the pinout

mck1117, 2023-01-23:
Switch inputs are practically free, add as many as there are pins on the MCU. 

Frequency inputs are very much NOT free. Flex is ok since it’s slow. 4 wheel speeds is too much.

rusefillc, 2023-01-30:
"4 wheel speeds" makes me think secondary F4 MCU

ElDominio, 2023-01-30:
How many frequency inputs can we have? At least a ballpark, so 3x crank/cam and 4x VSS is not possible? 

Still a lot more switch inputs are good (ac req, power steer cramp, oil press, etc etc

closed 2023-07-23 by ElDominio