rusefi/alphax-8chan #65: Pins without description, safe to assume the function under " 8chan is working" ?

ElDominio, 2023-01-17:
refers to https://docs.google.com/spreadsheets/d/1-h8c47vzwoVt56TxdpPsfDbb4WIOS3K4TfcJ-AowHh8/edit#gid=0

some pins have no comment at all, leaving their function vauge. Are they NC or does the assigned function under 8chan work?

ElDominio, 2023-01-29:
@rusefillc

rusefillc, 2023-01-29:
I have no idea what is " 8chan is working" but I have a feeling that the question here is "is https://rusefi.com/docs/pinouts/hellen/alphax-8chan/ done"

Answer: no, https://rusefi.com/docs/pinouts/hellen/alphax-8chan/ is not done

@ElDominio are there any specific pins you need available on https://rusefi.com/docs/pinouts/hellen/alphax-8chan/ before others?

rusefillc, 2023-01-30:
#45

closed 2023-07-23 by ElDominio