rusefi/alphax-8chan #66: OUT_DC1/2/3 can be used as aux?

ElDominio, 2023-01-17:
Them being called "low side flyback" leads me to believe I can use them all to trigger something with ground at the same time?

Is there any restriction (IE : using DC- as ground trigger only allows DC+ to be used as positive trigger OR cannot be used at all)

rusefillc, 2023-01-22:
Let me start with good news: DC2 and DC3 are totally here to be used as aux outputs, either low side or high side (with some limitation)

More about that at https://github.com/rusefi/alphax-8chan/issues/13

Now other news: I have no idea what would be needed on the software side. @mck1117 any chance you could help with an explanation and/or software if such would be needed? At least the explanation to begin with I guess :)

ElDominio, 2023-07-23:
Idk and with current YAML, i think having these pins dedicated to DC motor control is fine. Being able to be used high/low is a bonus

closed 2023-07-23 by ElDominio