rusefi/alphax-8chan #68: How does VR work?

ElDominio, 2023-01-29:
Example, Coyote engine has VR on crank and two cams (other cams are Hall), how do I wire it?

rusefillc, 2023-01-30:
I believe it's only two VR on 8chan at the moment

[image]

ElDominio, 2023-01-30:
I need one more :(

ElDominio, 2023-05-05:
Where do I wire VR-? @rusefillc

rusefillc, 2023-05-05:
@ElDominio Analog Ground https://rusefi.com/docs/pinouts/hellen/alphax-8chan/?pin=24A

ElDominio, 2023-05-05:
[8chanvrcoyote.zip](https://github.com/rusefi/alphax-8chan/files/11410621/8chanvrcoyote.zip)

Is this okay??
[image]

It might not have worked because I am dumb

closed 2023-07-23 by ElDominio