rusefi/alphax-8chan #76: 23A is it hall1 or cam_hall or else? very confusing

rusefillc, 2023-03-11:
yaml says hall 1
```
  - pin: 23A
    function: Hall Input 1
    # IN_CAM
    id: A6
    class: event_inputs
    ts_name: 23A - Hall Input 1

```

schematics says IN_CAM_HALL

[image]

schematics also has HALL1 which seems to be part of configurable something

NEED COMMENTS on the schematics

andreika-git, 2023-03-11:
On the schematics:
HALL1 is a configurable part of IN_CRANK with switchable pull-up.
CAM_HALL is a generic digital input with configurable pull-up/pull-down feature.
HALL2, HALL3, HALL4, FLEX are generic digital inputs with permanent +5V pull-up.

rusefillc, 2023-03-13:
@andreika-git 
``HALL1 (8A) is a configurable part of IN_CRANK with switchable pull-up.``

what does this mean? what is configurable between what and what?

Qwerty-OFF, 2023-07-20:
Pulldown/Pullup
@rusefillc should I do something?