rusefi/alphax-8chan #81: hall3 LED blinks but counter does not go up

rusefillc, 2023-03-13:


rusefillc, 2023-03-13:
Fun fact Hall4 5c which does not have LED works from software perspective https://github.com/rusefi/alphax-8chan/issues/77

rusefillc, 2023-03-13:
Rumors are same for Hall2?

rusefillc, 2023-03-13:
J17 works good

J17 is 5v when 31A is floating
J17 is 0v when 31A is ground

rusefillc, 2023-03-13:
[image]

```
  - pin: 31A
    function: Hall Input 3
    # IN_DIG11
    id: E15
    class: event_inputs
    ts_name: 31A - Hall Input 3
```

rusefillc, 2023-03-13:
[image]

rusefillc, 2023-03-13:
[image]

rusefillc, 2023-03-13:
Adding R125 has fixes two boards in NJ

@mck1117 FYI since you have a board

@ElDominio your board would need repair https://github.com/rusefi/alphax-8chan/issues/82

ElDominio, 2023-06-15:
R125 does not show in iBom @rusefillc

rusefillc, 2023-06-15:
@ElDominio see also https://github.com/rusefi/alphax-8chan/issues/82#issuecomment-1467066835