rusefi/alphax-8chan #82: 8chan needs R125 - fix hall2 and hall3

rusefillc, 2023-03-13:
@ElDominio when it would be good time please plan to ship your 8chan to NJ for R125 to be added, that would restore hall2 and hall3

ElDominio, 2023-03-13:
what is R125 (value?)? I might be able to do that one with good local board repair guy

rusefillc, 2023-03-13:
it's a quad 0602 4.7K resistor array https://www.mouser.com/ProductDetail/652-CAT16-471J4LF

same part as we were messing with on Hyundai but very different location - you might have a spare part from me or you might have ordered at some point

ElDominio, 2023-03-13:
ah fuck that i'd rather send it to you

rusefillc, 2023-03-25:
boards 2 & 3 fixed
@Timbelom please fix board 1

rusefillc, 2023-05-15:
Another relevant minor issue https://github.com/rusefi/alphax-8chan/issues/104