rusefi/alphax-8chan #100: CPU hangs if powered by USB

ElDominio, 2023-05-12:
The ECU hangs if i power it through USB. Normal connection procedure is key on, then connect USB cable. Sometimes, this also causes CPU to hang (blue LED does not blink)

If I press reset while USB is connected, the blue LED blinks for a second then dies (and no USB connection on laptop), I MUST disconnect USB, power off the car, power on the car, THEN connect USB to be able to connect.  This still errors out but not nearly as constant as powering ECU through USB, then keying on.

Bad procedure: 
1) USB connected
2) (nothing works)

Correct procedure:
1) Apply external 12v and gnd
2) connect USB
3) very low chance of crash (still crashes sometimes)

rusefillc, 2023-05-12:
@ElDominio please reset firmware to defaults - can you reproduce it?
@ElDominio please consider linking a tune

ElDominio, 2023-05-12:
Tune at https://rusefi.com/online/view.php?msq=1469

I reset to defaults, still does the same thing