rusefi/alphax-8chan #99: Odd error when assigning injector 5?

ElDominio, 2023-05-12:
[image]

But I can't see that in the full pinout?

ElDominio, 2023-05-12:
also happens on injector 6
[image]

probably 7 and 8, will test if needed

rusefillc, 2023-05-12:
@andreika-git see https://github.com/rusefi/rusefi/commit/0fb88270bce9743b02e18be4a47ea28c5adc3300 do we need to add pull-up control lines? where are those?

rusefillc, 2023-05-12:
open question if fix breaks anything with configurable pull-ups so we need @andreika-git

closed 2023-05-14 by rusefillc