rusefi/alphax-8chan #106: headers 3D

rusefillc, 2023-05-31:
3D seems to be missing ECU headers which are useful for case design

слева TE 6473423-1 https://www.te.com/usa-en/product-6473423-1.html
посередине 6437288-2
справа 6437288-3

[image]

rusefillc, 2023-05-31:
@Ringenium первый шаг - положить step в https://github.com/rusefi/kicad6-libraries?

Ringenium, 2023-06-01:
@rusefillc добавил степы

rusefillc, 2023-06-01:
@Ringenium вижу проресс библиотеки https://github.com/rusefi/kicad6-libraries/commit/db562ba9c1676e8cd117dd0bab22ead44a21c2a2

но ещё надо https://github.com/rusefi/alphax-8chan в самой плате как-то помочь? или прямо футпринт нужно связать с 3D?

Ringenium, 2023-06-01:
@rusefillc понял, сегодня поправлю и на плате

Ringenium, 2023-06-03:
[image]
подгрузил 3D файлы разъемов, но это не панацея и каждому пользователю нужно самому указывать путь к 3D файлу. Либо чтобы каждый раз так не делать, нужно в пути указывать переменные окружения и класть 3D модели в папку с кикадовскими моделями, типо вот так 
[image]

rusefillc, 2023-06-15:
@Qwerty-OFF нужет твой совет

текущая проблема - 3D файлы скопированы

Qwerty-OFF, 2023-06-15:
Кажется работает такая схема:
- библиотека компонентов лежит субмодулем в папке проекта
- в библиотеке компонентов хранятся 3д модели
- в посадочном месте пусть прописан относительно
[image]

Я [обновил](https://github.com/rusefi/kicad6-libraries/commit/a94ebd3fa1f0d04625b9a04affa1ee210521cff5) библиотеку, проверьте меня

Ringenium, 2023-06-17:
@Qwerty-OFF У меня что-то не сработало( 
[image]

Qwerty-OFF, 2023-06-17:
@Ringenium надо было футпринты обновить в плате, чтобы изменения появились, делал? Я могу вечером сделать если нужно

Qwerty-OFF, 2023-06-19:
Возможно придется обновить кикад, потому что у меня свежий
Проект теперь выглядит вот так:
[image]
При этом в самом компоненте путь к модели относительный как на моей картинке выше.
@Ringenium убедись что у тебя теперь модельки тоже работают

Ringenium, 2023-06-20:
@Qwerty-OFF  а какая версия у тебя, 6?

Qwerty-OFF, 2023-06-20:
7

Ringenium, 2023-06-20:
@Qwerty-OFF Так можно в 7 работать, проблем с переносимостью проектов не будет?

Qwerty-OFF, 2023-06-20:
@Ringenium переезжать в 7 можно, просто потом в 6 эти проекты не откроются

Qwerty-OFF, 2023-07-19:
Кажется это можно закрыть

closed 2023-07-19 by Qwerty-OFF