rusefi/alphax-8chan #107: What component to remove to ECU does not stay powered with a constant 12V source on an output?

ElDominio, 2023-06-13:
Mustang alternator has a cable that always has 12v, when it is grounded the alternator starts charging. I need an output that can do this without backfeeding the ECU

Basically same as https://github.com/rusefi/alphax-4chan/issues/63

ElDominio, 2023-07-05:
@Qwerty-OFF  @andreika-git  @rusefillc

rusefillc, 2023-07-05:
@ElDominio can you please loudly state what pin are you trying to use?

@ElDominio please clarify if you had a chance to look at 8chan schematics PDF to see corresponding flyback diode.

Overall let's discuss how do we document flyback removal without asking humans every time such removal is desired? Shall we designate some outputs for relays versus some relays (without flyback) for solenoids (with flyback)?