rusefi/alphax-8chan #112: 8chan rev 2 yaml

rusefillc, 2023-07-22:
https://github.com/rusefi/rusefi/blob/master/firmware/config/boards/hellen/alphax-8chan/connectors/8chan-A-B.yaml
https://github.com/rusefi/rusefi/blob/master/firmware/config/boards/hellen/alphax-8chan/connectors/8chan-C.yaml
https://github.com/rusefi/rusefi/blob/master/firmware/config/boards/hellen/alphax-8chan/connectors/8chan-D.yaml

produce https://rusefi.com/docs/pinouts/hellen/alphax-8chan/

We would need to update.

rev A data could be completely disregarded

Qwerty-OFF, 2023-08-01:
C and D look unchanged

ElDominio, 2023-08-01:
I remember adding much nore to C and D, is my yaml merged?

Qwerty-OFF, 2023-08-01:
Please check
**HALL x7:**

- VSS
- HALL1
- HALL2
- HALL3 (PU/PD)
- HALL4 (PU/PD)
- HALL5 (PU/PD)
- FLEX (PU/PD)

**VR x4:**

- VR1 (VR/HALL) MAX9924
- VR2 discrete
- VR3 discrete
- VR4 discrete

**DC motor x5**

**CAN x2**

**SWITCH x4 (PU only, not configurable)**

**WBO x2**

**Analog (PD only, not configurable) x13:**

- PPS1
- PPS2
- TPS1
- TPS2
- AUX x9

**Analog TEMP x4:**

- IAT (PU/PD)
- CLT (PU/PD)
- AUX x2 (PU)

**INJ + IGN x12**

**LS x14:**

- VVT x4 (high-current VND10)
- PUMP_RELAY (low-current)
- FAN_RELAY (low-current)
- MAIN_RELAY (low-current)
- NOS (low-current)
- BOOST (high-current)
- AUX x5 (high-current)

**TACH LS/5V/12V**

**HS x3**

**KNOCK x2**

**VIGN**

ElDominio, 2023-08-01:
Please check
HALL x7:

    VSS (PU/PD)
    HALL1(PU/PD)
    HALL2 (PU/PD)
    HALL3 (PU/PD)
    HALL4 (PU/PD)
    HALL5 (PU/PD)
    FLEX (PU/PD)

VR x4:

    VR1 (VR/HALL) MAX9924
    VR2 discrete
    VR3 discrete
    VR4 discrete

DC motor  **x4**

CAN x2

SWITCH x4 (PU only, not configurable)

WBO x2

Analog (PD only, not configurable) x13:

    PPS1
    PPS2
    TPS1
    TPS2
    AUX x9

Analog TEMP x4:

    IAT (PU/float)
    CLT (PU/float)
    AUX x2 (PU)

INJ + IGN x12

LS x14:

    VVT x4 (high-current VND10)
    PUMP_RELAY (low-current)
    FAN_RELAY (low-current)
    MAIN_RELAY (low-current)
    NOS (low-current)
    BOOST (high-current)
    AUX x5 (high-current)

TACH LS/5V/12V

HS x3

KNOCK x2

VIGN

only noticwed a problem that it should be 4x DC, all else looks fine

Qwerty-OFF, 2023-08-01:
> VSS (PU/PD)
HALL1(PU/PD)
HALL2 (PU/PD)

Should all HALL have switchable pull-up?

> only noticwed a problem that it should be 4x DC, all else looks fine

I see ETB + 4x DC in yaml.

rusefillc, 2023-08-01:
```

  - pin: 7C
    function: ETB positive
    type: etb
    ts_name: 7C - ETB+
    color: purple
    
  - pin: 8C
    function: ETB negative
    type: etb
    ts_name: 7C - ETB-
    color: purple

  - pin: 26A
    function: DC1 positive
    type: etb
    color: purple

  - pin: 27A
    function: DC1 negative
    type: etb
    color: purple

  - pin: 28A
    function: DC2 positive
    type: etb
    color: purple

  - pin: 29A
    function: DC2 negative
    type: etb
    color: purple

  - pin: 25D
    function: DC4 positive
    type: etb
    color: purple

  - pin: 26D
    function: DC4 negative
    type: etb
    color: purple    

  - pin: 1D
    function: DC3 positive
    type: etb
    color: purple

  - pin: 2D
    function: DC3 negative
    type: etb
    color: purple

```

ElDominio, 2023-08-01:
That was a mistake on my part, I will update YAML to remove

And yes, selectable PU/PD on all hall inputs