rusefi/alphax-8chan #119: 8chan-B test log

rusefillc, 2023-09-25:
USB power

* makes high pitch not very loud noise
* no devices in device manager
* no "too much current on USB" windows pop-up

probably two blue LEDs on power supply
no LEDs on MCU module
a LOT of LEDs go ON. first moment those are brighter and noise is louder
then red ones get less bright and noise gets less loud

rusefillc, 2023-09-27:
[image]

rusefillc, 2023-09-28:
[image]
[image]

even with leads cut we get 0.6A and hot MCU corner

[image]

thirstyone, 2023-09-28:
to me it looks like it's u100, not mcu (which is a CAN transceiver???) (or it's just the IR image is shifted a bit?)

rusefillc, 2023-09-28:
> IR image is shifted a bit?
totally shifted. secondary hot spots are 3v regulator (note it's outside of board) and power supply, like 2-3cm shifter to left.

rusefillc, 2023-10-04:
By the way we still have continuity buzzing between GND and 3v

thirstyone, 2023-10-04:
If I were you, I would have taken the chip off the board - could easily be a shorting inside the chip...